HeronMkII/pcbs #11: Systems debug PCB should be migrated over to Altium

vogeldylan, 2018-06-25:
Existing layout and schematics have been done in KiCad. To match the rest of the schematics, the systems debug project should be moved to Altium as well.

LornaLan, 2018-06-26:
No problem, will start doing that soon

vogeldylan, 2018-07-12:
@LornaLan updates?

LornaLan, 2018-07-13:
@vogeldylan Hey Dylan sorry I've been pretty busy for the past two weeks, should be better in a few days or so. Is there a deadline for this migration task?

vogeldylan, 2018-07-13:
Well the final deadline for finishing the new drawings is August 18th. There will be some changes to the design (i.e. issue #9) which need to happen before then, so migrating the existing schematics should be done relatively soon.

No worries though, I know you're busy. Just take some time to get started so that you can get a feel for the software.

brunofalmeida, 2018-08-18:
@LornaLan You should start using GitHub for the systems PCB and commit your changes as you develop it