HeronMkII/pcbs #14: PAY-SSM v3 001 Manufacturing

vogeldylan, 2018-07-19:
First of the v3 PAY SSM boards, pretty exciting. This issue will track manufacturing notes and rework/repair checklists while the board is being assembled.

* Missing Parts*
- 1 FDC637AN N-MOS

* TOP SIDE:*
- [ ] C9 and C10 on CAN are shorted
- [ ] Pins 7 & 8 are shorted on U9 (PEX)
- [ ] U10 (PHT2) has two pins shorted to vias
- [ ] U1 (DAC) has pins 1 and 2 shorted
- [ ] U13 also has a via short on the right side
- [ ] U18 has a via short
- [ ] Pin 19 and 20 on U17 are shorted
- [ ] Pin 18 and 19 on motor 2 IC are shorted
- [ ] Lots of solder balls around humidity sensor footprint

vogeldylan, 2018-07-19:
Updates from Bruno's inspection, and Dylan's rework

- C9 and C10 are a defect. They have been reworked once.
- Pin 7 & 8 on U9 are part of the same net, not a defect.
- U10 pins are on the same net, not a defect.
- U1 (DAC) pins are a defect. They have been reworked once.
- U13 pins are on the same net, not a defect.
- U18 is not a defect, same net.
- Pin 19 and 20 on U17 is not a defect, same net
- Pin 18 & 19 on motor 2 IC is not a defect, same net
- Solder balls on footprint is a process indicator. Balls have been wiped off by hand and inspected.

**TODO:**
- [ ] Inspect C9 and C10 rework
- [ ] Inspect U1 (DAC) pin 1 & 2 rework

vogeldylan, 2018-07-20:
- [ ] Pads around U1 (DAC) need to have caps attached
- [ ] Big solder glob on S3

vogeldylan, 2018-07-21:
Jaden and I finished up everything but the connectors and the pressure sensor on the bottom of SSM. Still missing N-MOS

- [ ] U3 looks shorted on pins 11 and 12 (buffer)
- [ ] U19 has some solder paste which hasn't re-flown on pin 13
- [ ] Pins 17, 18, 5 on U19 are shorted to vias. Need to check

brunofalmeida, 2018-07-22:
Programming headers:
PAY-SSM (right side) - notch facing down
PAY-Optical (left side) - notch facing down

brunofalmeida, 2018-07-22:
PAY-SSM 32M1:
- fuse programming worked
- program upload worked
- UART RX and TX worked
- UART headers:
    - top 2 - RX
    - bottom 2 - TX
- PROG <-> RUN header - need to be in RUN mode for SSM_RST button and UART to work

brunofalmeida, 2018-07-22:
PAY-Optical 32M1:
- fuse programming worked
- program upload worked
- UART RX and TX worked
- UART headers (from 32M1's perspective):
    - DI - TX
    - CLK - RX
- need to disconnect RX when programming

vogeldylan, 2018-07-28:
* Soldered on the missing 100pf
- [ ] Still missing that N-Mos, was never ordered
- [ ] missing thermistors to test with
- [ ] missing cheap heaters to test with

brunofalmeida, 2018-07-29:
- PWR/RST LED lights up when RST button is pressed
    - next iteration should use an open-drain inverting hex buffer - LED will light up when the button is not pressed

brunofalmeida, 2018-07-30:
- ADC numbers channels from 0-11, but schematic numbers thermistors from 1-12
    - future schematics should number thermistors and other components from 0-11

brunofalmeida, 2018-08-03:
Motor testing updates:
- possibly some problem because the CONFIG value is latched when exiting SLEEP, but CONFIG is tied to VINT which is only valid (high) when SLEEP is disabled - possibly a race condition
- cut CONFIG wire on MOTOR 2 and connected green wire so we can choose 3.3V (indexer mode) or 0V (phase/enable mode)
- motor seems to only work in phase/enable mode (CONFIG low) - will edit library to use that mode
- need fast SPI transactions on port expander for switching - SPI clock is currently around 125kHz (division of 64 from clock frequency of 8 MHz), should increase SPI clock speed in lib-common

brunofalmeida, 2018-08-09:
Motor testing:
- broke the green wire on the motor 2 driver CONFIG pin, very little of pin left so difficult to solder
    - CONFIG is pulled down internally so did not connect anything to it - defaults to phase/enable mode
- implemented phase/enable mode code
  - works for motor 2 driver
  - nothing happens when connecting the motor to motor 1 driver, so it is assumed to properly be in indexer mode as per the original PCB design
  - should ideally control CONFIG pin in software for a future iteration
  - should account for phase/enable mode pin setup so 32M1 vs. PEX pin assignments are balanced (e.g. AENBL and BENBL should both be on the 32M1 or both be on the PEX)
- period of motor cycling is much longer than expected (for expected ~2s, got ~12s)
    - probably because of slow SPI clock speed, need to increase

siddMahen, 2018-08-09:
@brunofalmeida SPI clock speed is currently 1/64th `F_CPU`, so about 125 kHz. Does this fit the approx. times?

brunofalmeida, 2018-08-10:
Motor testing:
- desoldered the CONFIG pin on the motor 1 driver, left open (default to phase/enable mode)
- fixed shorting the nFAULT pin on motor 1 driver
- tested both motors actuating the platform simultaneously - works
- whichever motor is connected to the motor 2 driver has better torque and is harder to stall
  - tried switching the motors - consistent result that it depends on the motor driver, not the motor
- hypotheses:
  - APHASE is tied together for both motor drivers, but BPHASE is separate between motor drivers - BPHASE requires 2 SPI operations on the PEX which messes up timing and changes the phase offset
- SPI too slow - try increasing clock speed
- eliminate/cache PEX read result and just do write operation

vogeldylan, 2018-09-05:
Soldered on the last remaining MOSFET. Have some prototype heaters which need to be tested with. Closing this issue.

closed 2018-09-05 by vogeldylan