HeronMkII/pcbs #15: PAY Optical V2 001 Manufacturing

vogeldylan, 2018-07-20:
- [ ] 0.47 uF 0603 caps missing

brunofalmeida, 2018-07-20:
- [ ] possibly 1 mOhm 1210 resistor missing

vogeldylan, 2018-07-21:
@brunofalmeida we found it in the 0603 resistors pile.

brunofalmeida, 2018-07-22:
- Used wrong footprint on PCB design for AD8605 op amp (PCB has SOP-65P200X125-5N, ordered/can only order SOT23-5). Need to respec using a different op amp model with a footprint matching the PCB.
- Need 0.47 uF 0603 caps (at least 2 for front of board)

brunofalmeida, 2018-07-22:
Notes:
- Chose not to attach oscillating crystals - will use wave generated by 32M1
- Chose not to attach optional gain resistors

brunofalmeida, 2018-07-25:
Chose AD8613 SC70 to replace AD8605

Other replacements considered:
- matching package, pin layout, and 1 pA bias current
- many are SC70 packages, which seems compatible with SOP-5
https://www.mouser.ca/datasheet/2/348/bd5291g-e-475488.pdf
http://www.ti.com/lit/ds/symlink/lpv521.pdf
http://www.ti.com/lit/ds/symlink/opa2348.pdf
http://www.analog.com/media/en/technical-documentation/data-sheets/AD8613_8617_8619.pdf
http://www.analog.com/media/en/technical-documentation/data-sheets/AD8691_8692_8694.pdf
http://www.analog.com/media/en/technical-documentation/data-sheets/AD8500.pdf
http://www.onsemi.com/pub/Collateral/NCS2006-D.PDF
http://www.onsemi.com/pub/Collateral/NCS2008-D.PDF
http://www.onsemi.com/pub/Collateral/NCS2009-D.PDF
http://ww1.microchip.com/downloads/en/DeviceDoc/20005791B.pdf
http://ww1.microchip.com/downloads/en/DeviceDoc/21733j.pdf
http://ww1.microchip.com/downloads/en/DeviceDoc/21881e.pdf
http://ww1.microchip.com/downloads/en/DeviceDoc/20002324C.pdf
http://ww1.microchip.com/downloads/en/DeviceDoc/20002321C.pdf

vogeldylan, 2018-08-03:
Issue was discovered whereby the footprint for the 470nm LEDs included a via in the center which allowed the LEDs to poke through the board. However, the pad radius surrounding the hole was too wide, and would short to either pad of the LED as it was soldered. As a result, both pins of the LED would be shorted together.

A fix was prototyped and confirmed working whereby the legs of the LED can be bent down at a slight angle to raise the part above the metal ring.

- [ ] Desolder the soldered 470nm LEDs
- [ ] Bend all of the LED legs
- [ ] Solder the modified LEDs onto pay-optical

vogeldylan, 2018-08-19:
Soldered the bottom side of PAY-Optical (non-reflowed side) with Cindy

- [ ] Missing MCP6546 comparator for CH5 selection
- [ ] Missing 330 0402s
- [ ] Missing 10K 0402s

vogeldylan, 2018-08-19:
Inspection report with Jaden:

*Bottom side*

- [ ] R14 is missing (upper left, center left)
- [ ] R27 is missing (center right, bottom left)
- [ ] U15 is missing
- [ ] Resistor in UTAT logo is missing
- [ ] 330 near A2-1 is twisted, connection looks bad on top
- [ ] 330 near A2-3 is not very well connected
- [ ] R19 questionable wetting on the right side (center left)
- [ ] R17 doesn't look connected on the right (center left)
- [ ] C25 is a little crooked (center left) 
- [ ] C25 is a little crooked (center left) 
- [ ] 330 near A1-4 doesn't look connected
- [ ] 2 330s near A1-7 could use better wetting
- [ ] U12 pin 14, 15, 16 are all shorted (center bottom)
- [ ] U12 pin 7 and 8 (center bottom)
- [ ] U12 center top needs some work. 1,2,3 also 4,5 also 7,8 also 9,10 also 11,12 shorted
- [ ] R1 resistor array looks like it has poor connection
- [ ] U1 pin 5,6 also 10,11 also 21,22 are shorted
- [ ] U1 top connections are big bulky balls, questionable
- [ ] Capacitor below C14, looks like it needs better connection on the right
- [ ] 330 to the right of A4-1 has poor wetting on the left
- [ ] R6 (center right) poor wetting on the left
- [ ] R18 (center right) poor wetting on the left
- [ ] U12 (top right) shorts on 1,2 also 4,5 also 7,8 also 11-14
- [ ] U12 (bottom right) shorts on 2,3 also 7,8 also 11,12 also 13-16 
- [ ] U12 (furthers right) shorts on 1,2 also 4-6 also 7,8
- [ ] C44 (beside bottom right U12) wetting on the left is bad

vogeldylan, 2018-08-19:
- Update PAY-Optical schematic with smaller footprints for the multiplexers.
- Manufacture with covered vias

brunofalmeida, 2018-08-28:
Top side inspection notes (my initial inspection):
[ ] U4 in bottom left quadrant, top row, 2nd from left - slightly crooked
[ ] C17 and R10 in centre block, top row, right side - shorted together
[ ] U1 (in middle) - pins 3/4, 6/7, 9/10 shorted
[ ] Resistor in top left near connector - shorted to via (probably intentional)
[ ] C1 and C45 (left centre) - might be shorted
[ ] U6 (top left, left side) - pins 4/5, 7/8 shorted
[ ] U7 (top left) - pins 1/2 shorted
[ ] U12 (top left, right side) - pins 5/6, 7/8 shorted, pin 16 shorted to resistor above
[ ] U6 (bottom left, left side) - pins 7/8 shorted?
[ ] R10 (top right block, bottom row, right side) - not soldered to one pad, crooked
[ ] U6 (top right block, left side) - pins 10/11/12 shorted, pins 4/5 shorted, pin 8 shorted to resistor above?
[ ] U12 (top right block, right side) - pins 12/13 shorted
[ ] U6 (bottom right block, left side) - pins 3/4, 14/15 shorted
[ ] U12 (bottom right block, right side) - pins 9/10/11 shorted, pin 16 shorted to resistor above?
[ ] C20 (bottom right block, bottom row, 2nd from left) - crooked, too close to C21

Repeated blocks:
[ ] U4 - 2 pins (pins 2 and 3) shorted
[ ] C17, R10 crooked
[ ] U12 - pins 7/8 shorted

brunofalmeida, 2018-09-01:
- R14 missing is not a defect, it's by design

vogeldylan, 2018-09-07:
Did a re-inspection of both sides, didn't find any of the issues you mentioned @brunofalmeida and didn't find any further ones.

brunofalmeida, 2018-09-07:
@vogeldylan I think this is done besides your notes for the next PAY-Optical iteration.