HeronMkII/pcbs #23: OBC EPS V1 001

jadenreimer, 2018-07-28:
- [ ] Real time clock is troubled. Remove and resolder
- [ ] C17 has poor wetting on both sides
- [ ] OBC 32M1: pins 17, 18, 19, 29 & 32 have poor wetting
- [ ] Top right of ABM3BF7W has poor wetting
- [ ] OBC MEM3: pins 1, 2, 3, 6, 7 & 8 seem unconnected with sad heel fillets
- [ ] R33: poor left connection to pad
- [ ] OBC MEM2: poor heel fillets on pins 1, 2, 7 & 8
- [ ] OBC MEM1: pins 1, 2, 3, 4 & 8 have poor heel fillets and pins 1, 3, 4, 6 & 8 have poor wetting
- [ ] Overall the wetting and heel fillets of the flash mems are questionable
- [ ] U7: poor wetting on pins 8, 12 & 14, missing heel fillets on pins 9, 11 &13
- [ ] EPS CAN: poor wetting & fillets on pins 1, 2 & 5
- [ ] OBC CAN: poor wetting on pins 1 & 8
- [ ] R31 is pulling a wheelie. Really big glob under one side
- [ ] C36?(the one in between R35 & R38: pretty spikey
- [ ] Overall cleaning of the board is in order

alihaydaroglu, 2018-07-31:
nice job on the inspection @jadenreimer good luck fixing my mess lol

alihaydaroglu, 2018-07-31:
adding @jackberezny here so he can do functional testing on it after it's been reworked

jadenreimer, 2018-08-04:
Soldering should all be fixed, we'll see how it turns out with testing. @jackberezny feel free to call me over tomorrow if something isn't cooperating and we can work it out

vogeldylan, 2018-08-20:
Assuming this is finished, the issue can be closed.

closed 2018-08-20 by vogeldylan