HeronMkII/pcbs #29: OBC PCB iteration

brunofalmeida, 2018-08-18:


vogeldylan, 2018-08-23:
Reference #4 #22 and outcomes from #26

vogeldylan, 2018-08-23:
Waiting on #27

brunofalmeida, 2018-09-30:
@hirolemo 

- Replace 32M1 and CAN transceiver sheets with the ones from pcbs-common (see the EPS board for an example of how to do it)
- Why is the RTC's VBAT connected to GND? We still need to add the backup battery right?
- Should number the flash memories 0-2 instead of 1-3
- Let's talk on Tuesday to clean up the file names

brunofalmeida, 2018-10-20:
@hirolemo note the new EPS-OBC header connections in the systems wiring diagram v0.5

brunofalmeida, 2019-01-02:
A couple of things:
1. Need to implement the open-drain buffer you selected for the RTC RST, I think this will take the last GPIO pin on the 32M1 so LED1 will be renamed
2. It would be easier for testing/debugging if the PGM <-> RUN switch was facing outwards (but make sure mechanically the switch doesn't extend past the edge of the board)
3. Silkscreen - need to label RST button, PGM <-> RUN switch, TX/RX for UART

brunofalmeida, 2019-01-04:
### Review - Part 1

1. See my last comment, points 2 and 3

2. PGM label is overlapping (there are other labels overlapping or upside-down but I assume this was intentionally left to deal with later)
[image]

brunofalmeida, 2019-01-04:
### Review - Part 2

1. Can we have a 5V pin on the debugging header? Would be useful for debugging I2C if it turns out to be 5V. I'm not sure if it would be better to replace something or add an extra pin. @vogeldylan what do you think?
[image]

2. Label the RST button
[image]

3. RX and TX text should be on the same side (right or left)
[image]

4. I'm confused about the LED labelling here (UHF and TUHF). This is supposed to be for TRANS_EN_UHF and TRANS_EN_UHF_OPT right? Maybe something like UHF and UHF_OPT?
[image]

vogeldylan, 2019-01-05:
See f99734c9bcc0b62879bd54d5f71784b96560c67c

closed 2019-01-05 by vogeldylan