HeronMkII/pcbs #31: Systems PCB iteration

brunofalmeida, 2018-08-23:
Notes for the new iteration of the systems PCB.

brunofalmeida, 2018-08-23:
My initial inspection and suggestions:
- should move files to a new `systems` folder in the root of the repo
- payload connections - PYL should be renamed to PAY for consistency
  - PYL -> PAY
  - PYL_RST -> PAY_RST
- UART pins - shorten names for visible labels? (not net names)
  - e.g. RX and TX instead of O_RX and O_TX if they are in the proximity of the OBC label
- rectangular outlines around sections of related components?
- numbers in top-centre near P1 and P2 are upside-down
- do we need resistor termination in specific locations for CANH and CANL?
- label which CAN pins go to which board?
- move coordinate reference point?
- align UART pins with programming headers and reset switches by board?
- label header pins
- more ground debugging headers?
- check placement of DSUB-25 header - probably needs to be closer to the edge

LornaLan, 2018-08-25:
Hey Bruno, thank you for your suggestion!

- should move files to a new systems folder in the root of the repo
done
- payload connections - PYL should be renamed to PAY for consistency
fixed
- UART pins - shorten names for visible labels? (not net names)
fixed
- rectangular outlines around sections of related components?
done
- numbers in top-centre near P1 and P2 are upside-down
fixed
- do we need resistor termination in specific locations for CANH and CANL?
not sure, can discuss
- label which CAN pins go to which board?
the CAN pins signals all come from OBC (the D-sub header)
- move coordinate reference point?
done
- align UART pins with programming headers and reset switches by board?
This is difficult, since the traces around the programming headers are already complicated, also UART pins are not really connected to them, they can be viewed as a separate part from the headers
- label header pins
I assume you mean P1, P2 and programming header pins? The space is a bit tight, should I label them on the Top overlay layer? (the space is tight for this, but could be done)
- more ground debugging headers?
Currently we have 6, 4 on P1 and 2 on P2. If we need more, how many do you think is suitable?
- check placement of DSUB-25 header - probably needs to be closer to the edge
I suppose you mean that I should move the smaller retangular part of the D-sub header out of the board (to save space?), which could be done (I checked the debugging board of last generation, they did have part of the header out of the board). However, because the current board space is sufficient, I wonder if it is okay to keep it like this. Will move it if there is any inconvinience when using the current printed and assembled debugging board (the D-sub headers are on the same position)
- check 6-pin programming header layout - might need to fix
Current layout is based on Issue #9
- add/label power LEDs
labelled, not sure which one to add
- reset LEDs?
They are above the programming headers

**I just noticed that the default labels are on the Top Overlay layer, while I have some labels on the Top layer. Should I move them all to the Top Overlay layer?

brunofalmeida, 2018-08-26:
I'll have to take a look at the previous iteration board in the lab to check some of these points.

brunofalmeida, 2018-09-07:
Updates (will clarify in person):
- CAN resistor terminations - this will be handled on the individual subsystem PCBs, not the systems PCB
- which CAN pins go to which board - my mistake, this doesn't apply to the systems PCB
- UART headers should be aligned with the programming headers and reset buttons - this will require some re-routing but is very important for programming and debugging
- label the debugging headers along each of their sides - important for debugging
  - will probably need to move the headers farther from the edge
- placement of DSUB-25 header - needs to be closer to the edge
  - we had to cut about 8mm from the edge of the last board
  - I now see why the header was placed farther from the edge so the footprint was aligned with the edge, but the footprint should extend outside the board so the male connector fits
- put all labels on Top Overlay layer instead of Top layer
  - will also address some of my comments about missing labels
- might not need so many CANH and CANL header pins?
- more ground debugging header pins? TBD

LornaLan, 2018-09-12:
Please check the new PCB layout I updated. Just started routing and will be happy about any comments about the new version.

LornaLan, 2018-09-14:
Routing of the new board is finished (yay)! Two things:
1) Increase wire width for 5v line? Solar+ line? Batt+ line? (didn't get a response from Mitchell, will ask him again)
2) The 25pin D-sub header layout changed - two GND pins are moved farther from each other.

brunofalmeida, 2018-09-14:
Nicely done, looking a lot better! Mostly smaller changes at this point; some of my suggestions are debatable, and some should be checked with Dylan/Ali and others.

For your points:
1. Ask Mitchell, but generally traces for power lines need to be thicker to handle higher currents.
2. I'm not sure why that would happen, confirm with Dylan.

- DSUB-25 connector placement looks good, but we should confirm in the lab by measuring how much we had to remove from the current board
- We might replace the DSUB-25 connector with something else (Ali is looking into this)
- Do we have 3D CAD models for more of these components to see the 3D preview better? (check with Dylan)
- Confirm TX and RX are in the right orientation and do not need to be swapped (should be TX and RX from the microcontroller's perspective, i.e. `print()` calls go to TX)
- Do we need so many pins for SOLAR+ and BATT+ if they are directly connected?
- Should we have PGM <-> RUN switches on this board?
- Does it make sense for the programming/debug headers that MISO is common but MOSI is different for each microcontroller?

LornaLan, 2018-09-15:
Updated version now on Github.

- Mitchell got back to me on the current needed for SOLAR+ and BATT+ line (5.82A), the calculated trace width is 4.6 mm. I tried my best to fit the trace on the board, but the traces near the D-sub header did not get widened, wonder if that will cause a problem...
- That footprint is from a standard library right? -> Some footprints (eg. resistors, headers) are from the standard library/UTAT library, but the pushswitch and the buffer is from custom library. Maybe can merge it to the standard/UTAT library? 
- Need to move/reroute some vias so they don't overlap with pins/components -> Fixed
- Label the reset buttons -> Forgot to put a big square around each board's debugging session, now fixed
- Do we need open-drain inverting hex buffers instead of open-drain buffers? -> Dylan and I discuessed this problem last time, the connections are remade and should be fine now. Inverting buffers are not needed on this board.
- DSUB-25 connector placement looks good, but we should confirm in the lab by measuring how much we had to remove from the current board -> could double check, but based on the schematic it should be this much:

[image]

- We might replace the DSUB-25 connector with something else (Ali is looking into this) -> I think Ali said "no" in today's meeting, if need any change in the future we could do another iteration.
- Do we have 3D CAD models for more of these components to see the 3D preview better? (check with Dylan) -> Right now just the resistor's have it, but if this is necessary I can look up the models.
- Confirm TX and RX are in the right orientation and do not need to be swapped (should be TX and RX from the microcontroller's perspective, i.e. print() calls go to TX) -> Currently MOSI is TX and SCK is RX for each board
- Do we need so many pins for SOLAR+ and BATT+ if they are directly connected? -> Checked with Mitchell, should keep it
- Should we have PGM <-> RUN switches on this board? -> Let me know if we need these switches
- Does it make sense for the programming/debug headers that MISO is common but MOSI is different for each microcontroller? -> There was a long debate of this last term... and the final result is the current layout
- The CANH label for the debugging header on the right is cut off -> Thank you for spotting that, fixed
- Check if there is a required manufacturing margin around the edges for label (confirm with Dylan) -> Sure, but I think if Altium accepts it, it should be within the manufacturing margin

brunofalmeida, 2018-09-18:
- Check in the lab how much of the previous board was removed for the DSUB-25 header
- Check with Mitchell about the widths of the power traces near the header
- Check with Dylan about the DSUB-25 footprint
- Some pins are unrouted in the layout (maybe this was intentional while we're working out the placement)
- Should probably label the reset buttons as "RESET"
- Might use mini version of DSUB-25 (ask Ali)
- PGM <-> RUN switches - check with Dylan

alihaydaroglu, 2018-09-18:
I think we should stick with the existing DSUB-25 for cost cutting purposes - we can keep the current wire and add some strain relief, and worst case we can cut and re-splice it if connections become iffy

alihaydaroglu, 2018-09-18:
i already spoke to Lorna with suggestions about the width of power traces - i think it would be smart to route them from the top side of the connector and keep thicker traces all the way to the pins. it might get a little tight

brunofalmeida, 2018-09-18:
@alihaydaroglu is a mini DSUB-25 connector more expensive than the normal one?

alihaydaroglu, 2018-09-18:
yes, significantly. the cables especially

vogeldylan, 2018-09-18:
* Short sections of thin power traces (like discussed for the EPS converter boards) is OK. Probably impossible to route them all the way to the pin, although they could be routed closer than they are now.
* The DSUB 25 footprint looks OK. Did you have a specific concern Bruno?
* Lorna I can get you set up with PGM<->RUN switches, and with more 3D models for the board.
* Everything on the board should be 15 mil (0.015") from the edge of the board.

brunofalmeida, 2018-09-19:
- I measured the old systems board in the lab, we removed about 7mm from the edge for DSUB-25 so make sure the footprint extends at least 7mm off the PCB (plus some more margin)
- Could we just group the power traces together on adjacent pins and run thicker traces that way?
- Discussed DSUB-25 headers with Dylan and Ali, we are using normal DSUB-25 on the systems PCB and mini DSUB-25 on the EPS PCB (to save space)

vogeldylan, 2018-09-19:
* Grouping them isn't a bad idea, just make sure the changes get relayed to EPS

LornaLan, 2018-09-21:
Summing up the changes need to make next:

- The routing for thicker traces of power lines -> Will group SOLAR+ and BATT+ together to share a thicker traces (they have the same power) and will route them from the top of the D-sub header
- Double check for unrouted pins, make sure that there is no such pins beside the 3v3 and GND ones
- Set up the PGM<->RUN switches
- Label the reset buttons
- Double check the margins between components and D-sub header and the edge of the board
- Get more 3D models

Anything else? 
Thank you for all the suggestions and advices!

brunofalmeida, 2018-09-21:
We'll need to check that the systems PCB header pin layouts are consistent with the systems wiring diagram I'm working on.

brunofalmeida, 2018-09-30:
The DSUB-25 header is extending about 9 mm off the edge of the board, should be good

LornaLan, 2018-10-01:
- The DSUB-25 should be good then. The pin layouts are (very likely) final.
- Checked with Mitchell, technically the SOLAR+ and BATT+ may run at 5.82Amp, but the wire is too thick to fit anywhere (I tried!). This is probably the best I can do right now.
- Will aligned
- Checked with Dylan before that 3v3 and GND don't need to be routed as they are connected to the 3v3 and GND plane already. Will double check if there are other pins unrouted.

LornaLan, 2018-10-02:
-  Dylan finished the 3V3 and GND polygon pour (all connections) today, so the connections should be good now.
- Need more 3D models
- Check connections/schematics one more time, if correct then everything should be good to go :)

brunofalmeida, 2018-10-20:
@LornaLan see the changes in the systems wiring diagram v0.5, notably:
- +SOLAR doesn't go to the systems PCB, it's only a current source going to the batteries (voltage doesn't matter so it doesn't make sense to have it on the systems PCB)
- Use +PACK instead of +BATT (see note on top_level sheet)
- Added RTC_VBAT (RTC backup battery voltage) - add to header so we can probe the voltage
- Other DSUB-25 pin layout changes (@vogeldylan can you comment on arrangement and GND placement?)

LornaLan, 2018-12-07:
Updates: Changed the D-sub layout according to systems wiring diagram. Some issues need to check:

- Ground and PWR pour
- SOLAR+ wire width should be increased, but no space for now

brunofalmeida, 2018-12-07:
We might want to add I2C_SCL and I2C_SDA to the systems debug header, but is there anything that would be worth taking out?

vogeldylan, 2019-01-01:
## Systems PCB Review Part 1

1. `CUSTOM.SchLib`, `system_debug_pcb.OutJob`, `Conn_Molex_39-30-1020_eec.SchLib` and `Conn_Molex_39-30-1020_eec_0.PcbLib` could not be found upon opening the project. The molex connectors should already be in CONNS.PcbLib as design item ID _WM1351-ND_. Please use the heron-general libraries, or integrate any additional components you require into these libraries. It will avoid any problems like this.
2. Your panel connectors look correct.
3. Your battery connector looks correct, just remember to label +ve and -ve terminals on PCB silkscreen
4. Don't see any updates to PCB on your commit. Please push or add the changes to the PCB.

vogeldylan, 2019-01-01:
@LornaLan

LornaLan, 2019-01-03:
1. Fixed the issue by organizing .schLib and .pcbLib to Heron General (will merge them into one single library soon after) with the latest commit
2, 3 Thanks! Should be labelled already.
4. Fixed.

vogeldylan, 2019-01-03:
## Systems PCB Review Part 2

1. More board needed for the connector; I went in and pushed it back.
[image]
2. You seem to be missing all of your GND and 3V3 vias on the board, so I added them.
[image]
3. Changed routing of PANEL+X+ and PANEL+X- to large polygons on VCC and GND layers, since the existing trace width was too thin. Also changed BATT nets to polygons.
4. Imported UTAT SS Rules file, re-enabled all of your batch DRC checks that were disabled and re-ran DRC.
5. Fixed outstanding DRC errors.
6. Otherwise, board looks good. Should be ready to send out.

brunofalmeida, 2019-01-03:
### My Review - Part 1

1. Can the programming headers have a notch to indicate the orientation of the programming header?
[image]

2. What does "RTC" mean in the top left debugging header? I think this isn't clear and should be more specific.
[image]

3. Can you flip the directions of the numbers and P1 in the debugging header?
[image]

LornaLan, 2019-01-04:
For @vogeldylan:
2. Why are Vias still needed there when we have board size ground/vcc pour?
Thank you for all the fixes!

For @brunofalmeida:
1. Which one is the right orientation? The notch on 2-6 or 1-5 side?
2. Changed to RTC-BATT, is it better?
3. Fixed

vogeldylan, 2019-01-04:
@lornalan vias are needed because the GND and VCC pours are on internal layers. To connect the top-side to GND and VCC requires dropping vias down.

brunofalmeida, 2019-01-04:
@LornaLan 
1. The notch should be on the 1-5 side, see #48. 
2. I think RTC_BATT would be better than RTC-BATT. I first thought RTC-BATT meant negative BATT according to the naming convention, so I think using an underscore would be clearer. It's already much better than RTC though.

### My Review - Part 2

(I know these are mostly small aesthetic changes)

1. Why do we need two separate LEDs for PANEL+ and PANEL- ? Is PANEL- just connected to GND or am I just confused on this?
[image]

2. Is part of the "BATT PWR" label going to be covered up by the connector below? Might want to rename "BATT PWR" to "BATT"?
[image]

3. I suggest renaming SOLAR+ to +SOLAR to match +PACK and the systems wiring diagram. That way it's more consistent with how you've structured the board labels (i.e. +PACK and +SOLAR for EPS monitoring, PANEL+ and BATT+ for "simulated" solar and battery signals)
[image]
[image]

4. Might want to move the RUN and PGM labels on the far right (for PAY) a bit closer to the switch to match the others.
[image]

brunofalmeida, 2019-01-04:
So the PANEL- LED should be removed?

vogeldylan, 2019-01-05:
Looks good d8ca9a4378e3418f5c5fd697efb40c93b7860bde

closed 2019-01-05 by vogeldylan