HeronMkII/pcbs #36: EPS PCB iteration

brunofalmeida, 2018-09-01:


brunofalmeida, 2018-09-30:
@teemoyeung @jadenreimer 
- In bt_current_monitor, what is the port connected to OUT supposed to be called? It looks like you have "0" instead of "O", and I don't understand the naming scheme
- For the PC104 headers, are "RxD" and "TxD" labelled from the OBC's perspective or the transceiver's perspective? Check the transceiver's datasheet and indicate in the schematic with a note
- Check current monitoring circuits with @mitchell (I think the orientation was flipped last time?)
- In eps_pv, should move the duplicated sections (5-pin header + 3 diodes) to its own schematic sheet
- Remove flash memory chip - should only be on OBC
- Let's talk on Tuesday to clean up the file names

brunofalmeida, 2018-10-20:
@jadenreimer note the new EPS-OBC header connections in the systems wiring diagram v0.5

brunofalmeida, 2018-10-23:
Should take a look at the IMU pins - do we need pins such as WAKE, H_INTN, NRST?

vogeldylan, 2018-12-31:
## EPS Review Part 2 

1. Your CAN terminations are fine. Two 60 Ohms with CM filter cap for 120 Ohm nominal termination.
2. D25 connections look good.
3. RTC VBATT GND connection to deployment looks OK, not too much current on that trace anyways.
4. Mode select switch, programming header, RST, and RX/TX header look good
5. Add a parameter to the `U_eps_32m1` schematic symbol in `eps_digital.SchDoc` called _LED6_ and label it _DNP_. This should flow down to the LED monitoring schematic and tell us to not place LED6 during manufacturing. 
6. Tidy up the silkscreen around the LEDs, move the resistor in the middle of TXCAN and RXCAN 
[image]
7. As per the above image, you have three vias on the left of your 5V trace but only one connected on the right. Drag out that trace.
8. Like I mentioned, you can take out all the jumper traces between adjacent butterfly pads. They're internally connected 
[image]
9. Heater control, 5V routing to transciever looks better than before. Just fix up that via problem mentioned in 7.
10. Change the tolerances of the 10K's attached to your thermistors from 1% to 0.1% just in case we forget later.
11. Still have yet to tie together GND underneath your boost converter
12. You have a net antennae on your buckboost GND
13. Perhaps not the end of the world, but ignoring the manufacturer power supply/layout recommendations is ballsy. The second image I linked shows one of the INA's which should be alright. Nice short traces between the sense resistor and INA, although lack of bypassing could always be an issue. The third one is the battery current monitoring circuit, which looks like it might benefit from some proper attention and sense resistor connections.
[image] 
[image]
[image]
14. I think I mentioned this last year, but I don't buy your current monitoring setup. I'm pretty sure for the direction of current you're trying to measure, you need to connect REF to GND, not 3V3. Take another look at the datasheet and make sure. Also, bypassing is a good idea. Also, your battery current monitoring will only monitor current going into the battery, i.e. charging. In this case, connecting REF to 3V3 is OK, but it forces you to only measure 'negative' current. Connecting to a resistor-divider at midscale would let you measure both directions. Also, depending on what ADC range you're using, you might want to connect REF on this to 2V5, if you're using the 2.5V range on the ADC
[image] 
[image]
15. While remembering year-old concerns I had with EPS, I also don't trust your buffer connections either. There are quite a few concerns I have here so I'll break them up into separate points.
16. Your +XVGS, +YVGS, etc. connections are coming from the port expander. The port expander outputs are defualt input, i.e. high impedance. Throw a pull-down resistor on those inputs to ensure that the shunts are default OFF to ensure we can turn on the satellite. 
17. You've connected the ADA4891 op-amp quad pack in a unity gain buffer configuration, which will mirror the 3V3 output of the port expander. I believe you want these to be in a comparator configuration to drive the gates either at 0V or 5V. Now, as shown on the datasheet, this won't really make a difference because the Rdson of the MOSFETs are so low (the I_D scale on the left is quite large) but just an FYI
[image]
18. The bottom ADA4891 you have mirroring the INA214 current monitors is connected in a ... I don't know what configuration. With the inputs tied together I think you'll get 0V output no matter what your current is doing. You should connect this one in unity-gain configuration.

TBC in Part 3 ...

brunofalmeida, 2019-01-01:
My EPS Review:
1. It would be easier for debugging if the PROG <-> RUN switch is facing outwards
[image]
2. You shouldn't name your designators with circuit specific names (e.g. Coutbb1), they should be a generic letter and number (e.g. C17). These names for your converter LRC components probably have some good use, but perhaps put them as the component comment or as generic text in the schematics.
[image]
3. Why is this cap here
[image]
4. Small point: if possible, try not to have your part designators overlap with the vias, because when manufactured, part of the number will be cut off
[image]

brunofalmeida, 2019-01-03:
### My review part 2:

1. Check all INA214 ICs (eps_bt, eps_batt, eps_bb): should REF be tied to GND for unipolar operation? Should +V be 2.5V?

2. eps_bb.SchDoc: Are the conenctions to IN+ and IN- correct for the current direction? The positioning of the GND port connected to pin 2 is confusing, kind of looks like the wire below is labelled with a GND net.
[image]

3. Is RTC_BATT_GND properly connected to the main GND? See bus-main -> eps -> eps_batt

4. Missing +SOLAR connection on pin 21 of header to systems PCB (according to systems wiring diagram) - is this intentional? Was a decision made on whether monitoring solar was feasible?
[image]

5. Any plans for a debugging header?

6. Can you make the TX and RX labels horizontal (to match the UART label)?
[image]

7. Can you put the PROG label below the header to match the other controls?
[image]

8. Maybe a single dash between the arrows here to match "PROG <-> RUN"?
[image]

9. In general, clean up the notes and questions in schematics that have already been dealt with.

brunofalmeida, 2019-01-04:
Update on INA214 current monitors: need to connect REF to GND, V+ to 3.3V (according to datasheet, must be between 2.7V and 26V)

vogeldylan, 2019-01-05:
Looks good d8ca9a4378e3418f5c5fd697efb40c93b7860bde

closed 2019-01-05 by vogeldylan