HeronMkII/pcbs #54: pcbs-common Enhancement and Implementation

jadenreimer, 2018-09-28:
List of things to change in pcbs-common design source files listed in the comments.

### **How to add pcbs-common files into your current Altium project:**
**Get the files in your current project**
Project →  Add Existing to Project → navigate to your copy of pcbs-common → select relevant files for your design.
**Substitute your current subsheets with the pcbs-common versions**
Open properties of target subsheet (the green box) → find the label indicating from where it’s taking the current subsheet → select the ...  → navigate to the pcbs-common version in your project → select and OK that bad boy.
**Synchronize the lables to the subsheet with pcbs-common version**
Right click on the green box → sheet actions → synchronize sheet labels → select the pairs of corresponding sheet symbols → select the bottom arrow option (symbol of a beige box to a blue box [this pushes the version from pcbs-common onto the upper level]).
When you run out of things to link, select the remaining items in the middle column → add (bottom option) → add no ERCs (X symbols) to pins you are not currently using.

jadenreimer, 2018-09-28:
- [ ] pex: VCCs are pointed down :(
can't remember other items

brunofalmeida, 2018-09-30:
@vogeldylan 
- [ ] Does it make sense to repeat the component name in external ports? e.g. for the ADC schematic sheet, should the port be called `GPIO` instead of `ADC_GPIO` since from context it's known to belong to the ADC? When that sheet is incorporated in a higher level sheet, it should definitely have ADC in the name though.
- [ ] Typo in adc-circit-ADS7952 ("circuit")
- [ ] In adc-circit-ADS7952, recommended input impedance is < XX?

closed 2019-06-23 by brunofalmeida