HeronMkII/pcbs #65: Consider using 64M1 microcontrollers

brunofalmeida, 2018-10-24:
We might want to use the 64M1 model instead of the 32M1 for more program memory. They have the same datasheet and seem to be the same in every other way, just that the 64M1 has more program memory, EEPROM, and SRAM.

alihaydaroglu, 2018-11-10:
this seems like a good idea... did we ever make a decision on it?

brunofalmeida, 2018-11-11:
Yeah, it should just be a matter of updating the flags in the makefiles.

hirolemo, 2018-11-21:
Are the 64M1s larger/smaller than the 32M1s. Has using them been decided?

brunofalmeida, 2018-11-22:
We decided to use the 64M1. It's exactly the same physical size and footprint as the 32M1 so that doesn't matter for PCB layout.

vogeldylan, 2018-12-31:
When the DigiKey order goes out I'll try to make sure the 64M1's are ordered.