HeronMkII/pcbs #71: OBC v2 Manufacturing

vogeldylan, 2019-01-19:
Initial issue creation

- [ ] Missing SPI-I2C bridge IC

brunofalmeida, 2019-02-06:
@hirolemo 's inspection report:
- [ ] U3 - pins 7-8, 11-13, 17-19, 31-32 shorted
- [ ] S2 - bottom left pin floating
- [ ] R6 - solder mount

brunofalmeida, 2019-02-06:
My fixes:
- [ ] R15 (R40 in schematic export) - comment says 30k, but value says 10 - for lowpass filter, 10 is the correct value but 0 is also OK

brunofalmeida, 2019-02-07:
Started testing software:
- Uploaded a program successfully (after patching `avrdude.conf` to support the 64M1 and adjusting the makefile)
- UART TX and RX working (`uart_print` and `uart_echo` programs)
- Programming switch and RST button working

vogeldylan, 2019-02-08:
@brunofalmeida can you change notes to use the bullet point, rather than checkbox? Were the solder issues you mentioned fixed?

brunofalmeida, 2019-02-08:
@vogeldylan Yeah makes sense. All the soldering issues we were aware of were fixed.

brunofalmeida, 2019-04-24:
Changed I2C from 3.3V to 5V:
- Soldered U6 (level translator IC), R3 (replaced with 220k), R16/R23 (replaced with 5.1k)
- Desoldered R21/R22

For OBC on its own (not connected to anything), 5V I2C seems to be working (oscilloscope probing shows SCL and SDA are at 5V)

brunofalmeida, 2019-04-25:
Tested I2C with 5V level translator, all seems to be working

brunofalmeida, 2019-06-08:
Changed I2C back to 3V3 configuration:
- Desoldered U6, C3, R3, R16, R23
- Left R7, R8 on (pullups to 3V3)
- Soldered R21, R22

closed 2019-06-23 by brunofalmeida