HeronMkII/pcbs #72: EPS Improvements for Flight Model

jadenreimer, 2019-02-09:
Notes for flight model improvement:
- debugging header for debugging: dedicated 3V3, GND, etc. for probing
- way of testing and powering converters as completely separate components from the rest of the board + dedicated converter debugging headers/test pads?
- better current-path design/layout is a must
- forethought into how we will power and test the board must also be considered in the design
- inductors: footprints need to be 100% confidence validated and an inductor current-monitoring resistor is a must

Challenges/notes on current design:
- converters are really hard to debug considering they are QFNs
- pcbs-common seemed to create new challenges every week
- reset button is really close to headers and hard to get at with fingers
- current-path had to be retro-fitted to existing routing instead of taking priority

brunofalmeida, 2019-02-09:
Should we be able to supply 3V3 and 5V directly (bypassing the power converters) from a power supply for testing software?

jadenreimer, 2019-02-10:
Very true, was thinking it when adding the debug bullet point but never did, thx

brunofalmeida, 2019-02-22:
- Add a pullup resistor to the IMU's CS line

brunofalmeida, 2019-03-06:
- Resistor for measuring inductor current? (ask @aumitche)

brunofalmeida, 2019-03-10:
Need to fix battery current monitoring setup with INA214 - REF should be tied to GND, as specified in the unidirectional operation schematic.

[image]

[image]

brunofalmeida, 2019-03-27:
- Label headers to heaters and thermistors (including numbering)
- Possibly debugging headers (e.g. ADC measurements, SPI, CAN)

vogeldylan, 2019-04-12:
EPS and OBC MISO_A should be connected on the EPS board, to enable programming through the debugging cable.

brunofalmeida, 2019-04-20:
Capacitor on +PACK to prevent transient voltage drops when the shunts are activated?

brunofalmeida, 2019-05-10:
- Consider putting an extra cap between the MCU 3V3 power and ground (between pins 4 and 5, in parallel with the existing capacitance) - used 50uF in testing (worked), then 200uF (worked better)

alihaydaroglu, 2019-06-05:
- move the current monitor on the solar panels to before the shunt to be able to tell when satellite is in the sun

jadenreimer, 2019-06-08:
- measured values from the micro differ from input, and measure a lower voltage with higher current: with a supply at 4V and 0.61A the micro reads 3.8V at 0.7A

brunofalmeida, 2019-06-16:
From BUS integration test on 2019-06-15:
- Need to swap positions of H1 and H2 in layout
- Need shorter battery and dep switch connectors or else transceiver doesn't stack properly

brunofalmeida, 2019-06-22:
- Change battery holder to fit the CR2330 batteries we already have

brunofalmeida, 2019-06-23:
- [ ] See important notes from #76

brunofalmeida, 2019-06-26:
- [ ] Move the 6-pin antenna header (currently 4-pin but will be 6-pin on flight model) farther from the battery connectors so it's easier to access

brunofalmeida, 2019-07-13:
- [ ] Might want to move physical placement of RTC battery holder so the battery can be taken in/out without interfering with OBC mechanical mounting