HeronMkII/pcbs #77: SYS PCB flight model improvements

brunofalmeida, 2019-03-21:
- Fix reset button connections (ask @vogeldylan)

LornaLan, 2019-04-04:
- *make sure EPS mini D-sub and SYS D-sub are aligned.*
- Fix EPS programming header

LornaLan, 2019-04-20:
EPS programming header's issue fixed (on EPS).

LornaLan, 2019-05-01:
- [ ] Fix the D-sub header to agree with the system wiring diagram
- [ ] Fix reset button connections
- [ ] Fix panel +/- label

brunofalmeida, 2019-06-16:
Look into issues with LEDs turning on and flickering on/off randomly (from BUS integration test on 2019-06-15)
- Stray charges?
- Pullup/pulldown resistors?
- More capacitors?

brunofalmeida, 2019-06-23:
- [ ] See notes from #70

brunofalmeida, 2019-07-31:
Need to fix battery connector - which footprint are we using?

LornaLan, 2019-08-22:
Used a screw terminal for battery connector, should be good

closed 2019-08-22 by LornaLan