HeronMkII/pcbs #81: Systems Wiring Diagram Update

jadenreimer, 2019-06-22:
- [ ] rearrange pins on the EPS PAY connector for a shorter path to converter outputs
- [ ] swap H1 and H2
- [ ] (potentially) change EPS-PAY connector power inputs (28V regulator case)
- [ ] 2 pins from OBC to EPS for release pins (if not already done)

brunofalmeida, 2019-07-06:
- Don't need to make any changes to H1 and H2 on the systems wiring diagram - this only applies to the EPS physical layout

brunofalmeida, 2019-07-07:
- [ ] Are there more things we could break out into debugging headers on the systems board? (e.g. I2C, more EPS measurements)
- [ ] Check pin 21 - make sure +SOLAR is changed to GND

jadenreimer, 2019-07-11:
- [ ] add RELEASE_A and RELEASE_B to EPS-SYS cable

brunofalmeida, 2019-07-19:
- [ ] Add a new signal called DIS_TRANS_UART (disable transceiver UART) to the EPS-SYS header
  - Might rename it active low (DIS_TRANS_UARTn) later