HeronMkII/pcbs #84: pcbs-common Improvements

jadenreimer, 2019-07-27:
- [ ] add a silkscreen notch for programming header orientation to footprint (does this matter with the change in programming header selection?)

brunofalmeida, 2019-08-15:
@vogeldylan In adc-circuit sheet, should VREF_2V5 be labelled as a power net instead of a signal net? It would clarify net naming if power nets take priority

vogeldylan, 2019-08-15:
@brunofalmeida can you give a use case where having 2V5 as a signal caused net naming issues? The signal isn't really meant to be used as a power rail (10 mA max, serves as the reference for the ADC so susceptible to noise)

brunofalmeida, 2019-08-15:
See ADC on current PAY-SSM design

vogeldylan, 2019-08-15:
I see your point but raise you the following 
[image]

vogeldylan, 2019-08-15:
(i.e. this shouldn't be an issue with this design, not sure why it is, not going to change the schematic in pcbs-common)

vogeldylan, 2019-08-15:
- Silkscreen notch should be added by the designer, since it isn't associated with the part itself
- CAN in our design doesn't require differential length matching

closed 2019-08-15 by vogeldylan

brunofalmeida, 2019-08-15:
@LornaLan I guess you need to tie the VCC2V5 output of the subsheet to a power net